EdgeTX/flasher #2: No way to see "Official Releases or Release Candidates" in flasher

phileville, 2021-06-20:
Currently in the flasher, the is no way to designate a "official release or release candidate".  In the current branching scheme, there is the 2.4 branch which has all the merges as well as the nightly branch which is everything that happened to be checked in at the time of the nightly run (I dont really see the benefit to having both of these branches - why choose the nightly over the most recent 2.4?).  In my opinion, there needs be an option for users to select the official release or release candidate, and an option to select a branch -  like 2.4 is now with all the recent past builds. 
Additionally, it would be great if somehow in the 2.4 branch for the autobuilds we could add the notes form the check in to the notes on the flasher so you would see what was changed - instead of just "Major initial bug fixes".  As it stands right now, there is no way for a normal user to download the RC1 with the flasher - you can only download the latest 2.4 build or an earlier one  - but there is no way to tell which one is the "release candidate".  I think that this needs to be addressed for the acutal release ideally.

CoderElectronics, 2021-06-23:
This Problems looks to be somewhat fixed, the Release candidate branches will be added as needed. Also in the latest release the commit id is now shown instead of the date stamp.

The Changelog section will need to be updated automatically, i can take a look at implementing this. I will get back to you shortly!

CoderElectronics, 2021-06-23:
@phileville The changelog section has now been changed to display commit messages for a changelog.

Here: https://github.com/EdgeTX/flasher/commit/9d29d24fd74f256f4858992902f01dd0c9e58870

closed 2021-06-23 by CoderElectronics

phileville, 2021-06-24:
This issue is still not closed in my opinion.  Please show me how to choose the RC2 in the flasher.  Normal users are not going to look through all the commit ids to find the RC.. Or am I missing something here
[image]

[image]