EdgeTX/flasher #3: Some rough edges still ;)

pfeerick, 2021-06-24:
First of all, it works! :) 🍕 🥳 

Thank you for taking on board the Transmitter model dropdown (rather than the zip name)... that makes it so much simpler! :)

Now, for the not so fun stuff...
Something is a bit iffy about the msg dialog and the time stuff is shown... 

I just ran through on my TX16S, with the transmitter in bootloader mode, on Windows 10. 

I had the explorer window open so I could see what happens, and after clicking the 'Download Package' button (perhaps that should be 'Write to SD Card'?) , I could see the folders being erased from the SD Card, nothing was showing on the Flasher yet. 

i.e. the dialog in the screenshot wasn't being shown, and the message marked as 1 wasn't being shown whilst that was happening. The dialog finally appeared after the SD card was complete erased (including MODELS and RADIO!?!) and was immediately at message 2. Message 3 wasn't shown until after that had happened. So it seems like messages are being shown *after* they are completed, not before or whilst they are running.

'Finished voicepack 0' wording is a little confusing... but the voicepack was installed. 

[image]

Since this erases the SD card, perhaps more emphasis on the warning?
https://github.com/EdgeTX/flasher/blob/c0f922f2038422e3376e7b063d1126915c1ca06e/src/components/SDCardPage.vue#L59

I'm not sure if this is right... I copied it from [here, but it's being used for list elements](https://www.codementor.io/@mtt/5-ways-to-change-component-colors-and-styles-using-vuetify-13oz2o0pt0#2-using-the-class-property)... hopefully give you some ideas ;)

`<p>Please select the SD Card you want to write to. <span class="red--text">Please note this will erase the contents of the selected disk, so make sure to select the correct one!</span></p>`

- [ ] Check why UI messages are showing after the task have completed (including not visual feedback until SD card erased)
- [ ] More emphasis on fact SD card will be erased
- [ ] Perhaps *don't* delete the `RADIO` and `MODELS` folders... I wasn't expecting to need to re-calibrate/re-configure transmitter and re-add models
- [ ] Tweak to the 'Finished voicepack 0' message ... "Finished installing Voicepack"?
- [ ] Maybe re-label button ... it's not really "Download Package" ... "Write to SD Card"?

I look forward to seeing this endeavor continuing to evolve and get even better! :)

pfeerick, 2021-06-24:
This is probably slow since I'm running a small card via the TX16S... but gives you a idea as to what's happening:
https://user-images.githubusercontent.com/5500713/123221099-ec446e00-d511-11eb-8371-fdac2a8289d4.mp4

phileville, 2021-06-24:
The other thing that I found a bit rough - if you connect the radio powered off, like you need to flash FW- then the flasher does not detect the SD card (naturally).  IF you connect via USB with the radio powered on, the flasher will write to the SD card the card contents, but cannot flash the FW, because it cannot find the DFU connection.  I understand why it is like this and maybe this is a limitation that just needs to be well documented, unless someone has and idea around that.

CoderElectronics, 2021-06-24:
@pfeerick @phileville I am adding a popup for warnings and instructions on the radio state. That should address the unclear messages. I will take a look and try to fix these today

CoderElectronics, 2021-06-24:
@pfeerick All adjustments you suggested should be in the latest commit: https://github.com/EdgeTX/flasher/commit/2f1427fdef489ea08eb88c314bc5c4e6eb8a9a05

pfeerick, 2021-06-25:
This is pretty much done, I have some more suggestions, but will do those as individual issues as they're in relation to more than just the sd card section.

closed 2021-06-25 by pfeerick