EdgeTX/flasher #4: Wrong SD card pack version...

pfeerick, 2021-06-24:
When I selected the TX16S as the target radio, it seems like the complete wrong version of the SD card pack was downloaded. It's not the zip file, but the wrong version - i.e.  background graphic is the opentx one, not the edgetx one and there are scripts which were removed from the pack a week or so ago). I double checked against the latest rolling release of the sdcard repo just to check I wasn't going crazy! 😆

CoderElectronics, 2021-06-24:
Ok so we need to diff and merge in changes from RC1, it is still functional but there are changes that for some reason are in RC1 instead of in the latest repo through a PR. Let's see if we can get all those into the master branch of the SD Card repo.

pfeerick, 2021-06-25:
No, not quite - it _seems_ like there is a recursion problem when the files are written to the SD card. 

More here:
https://discord.com/channels/839849772864503828/839865370440630302/857779432454225961

CoderElectronics, 2021-06-25:
I believe your PR fixed the issue, thank you!

closed 2021-06-25 by CoderElectronics