EdgeTX/flasher #5: Check for a valid target before showing erasure warning...

pfeerick, 2021-06-25:
... as if you haven't picked a valid target, you're not going to be erasing anything...

i.e. current behaviour: 
[image]

pfeerick, 2021-06-25:
Also, it would probably be a good idea to prompt to user to take the transmitter out of DFU mode if they have just updated the firmware.

pfeerick, 2021-06-25:
i.e. "Please select a valid disk, and ensure your transmitter is not in DFU mode"

CoderElectronics, 2021-06-25:
Should be all fixed, please take a look

CoderElectronics, 2021-06-25:
The dialog display order is fixed here: https://github.com/EdgeTX/flasher/commit/8d5b14a8e7e8186dbe9344dfc4ca2015740cfd6c

closed 2021-06-25 by CoderElectronics