EdgeTX/flasher #10: [SDCARD] SD cards with labels (possibly Linux specific)

pfeerick, 2021-06-26:
I think a side effect of this test is that SD cards that haven't had a label set won't appear in the drop down? I can understand the mounted part, that's part and parcel of doing stuff on Linux, but *requiring* a label is going to be annoying unless the requirement is clearly stated. i.e. I never bothered setting a volume label for the SD that came with my T-Lite, and it didn't appear in the list - so I just had to set one after reviewing the code to work out *why* it didn't show. I can't say for certain I noticed the problem on Windows - I'll have to try again with another unlabelled card to check. 

https://github.com/EdgeTX/flasher/blob/879a7159b4371cf50c3d602e21fce847fedbdb08/src/components/SDCardPage.vue#L263

pfeerick, 2021-06-26:
I think this is a duplicate of #10

pfeerick, 2021-06-27:
Just verified it's also happening on Windows, so it will be because of that filter. It may be better to drop that requirement, and possibly also add the size of the drive to the dropdown, thus if a system drive is included in the list by accident (which _generally_ shouldn't because of the removable requirement, the end user can eliminate by size.

CoderElectronics, 2021-07-01:
The filter has been removed so it should work fine now, please re-open if it wasn't solved.

closed 2021-07-01 by CoderElectronics