EdgeTX/flasher #13: False Positive in AntiVirus Software

jlevy-dev, 2021-07-01:
Flasher .exe file has been coming up with a false positive as a trojan as it seems to be a sub package that is causing it to get flagged.

[Link to Virus Total Checker]( https://www.virustotal.com/gui/file/3c095a4f5fe73e46563b56ed41bf9c705ac70eb958c9950ba35568f31fb169c9/detection)

There have been other instances of this in other programs link [here](https://github.com/megai2/d912pxy/issues/284) with their fix to this. And [here](https://github.com/AvaloniaUI/Avalonia/issues/5937) as well. 

[image]
Image credit[ RC Video Reviews](https://www.youtube.com/user/jmxp69)

[image]
Image on my windows defender

jmxp69, 2021-07-01:
Disabling Windows Defender allows the file. Same issue occurs on MS Edge.

pfeerick, 2021-07-01:
Probably varies from build to build of the flasher / or there has been enough reporting already to get rid of the false positive, as the latest build does not get flagged. https://www.virustotal.com/gui/file/450bcc0d2b4453bdd02659ac4846a922e75d3b36663563c6e577b3e69041d5a7/detection

jmxp69, 2021-07-01:
I just downloaded it and it got flagged by defender.

jmxp69, 2021-07-01:
[image]

pfeerick, 2021-07-01:
OFFS Microsoft... between one scan by VirusTotal, and a second one an hour or so later, it is being picked up again by Microsoft Defender *only*. It also depends on what version of the 'security intelligence database you have... 1.343.151.0 didn't flag it, but about two hours later, and 1.343.155.0 does. 

Defender can't make it's mind up... the same file is either "Trojan:Win32/FormBook.AVP!MTB" or "Trojan:Win32/Emali.A!cl" depending which time it sees the file.

jlevy-dev, 2021-07-02:
My understanding of this is that edge tx needs to make a request to Microsoft to be removed from the black list. I’ll look around for details on how to do that.

jmxp69, 2021-07-02:
It's already been taken care of Justin. I think they can close this one now. Latest definition file allows the flasher.

pfeerick, 2021-07-02:
@jlevy-dev Feel free to close the issue if that latest Windows Defender update definitions update resolves this for you. For future reference, this is the starting point in reporting false positives (or submitting files that have malicious content but are not being detected, for identification)... Even though rather confusingly it's in relation to "Microsoft 365" - https://docs.microsoft.com/en-us/windows/security/threat-protection/intelligence/developer-resources

jlevy-dev, 2021-07-02:
Awesome thank you! Y'all rock!

closed 2021-07-02 by jlevy-dev