EdgeTX/flasher #19: MacOS Flash error

mkintzel, 2021-08-06:
Trying to Flash a RadioMaster TX16s from a MacBook macOS BigSur.  I have flashed this radio previously with OpenTX from OpenTX companion.  I have not tried that method for EdgeTX as was hoping to use your new utilities.  I receive the following message immediately upon pressing the "Flash Radio Firmware" button.  (By the way, this differs slightly from the Installation Guide which mentions selecting "Write")  Thank you for all your work on this project!!
[image]

artem-alexandrov, 2021-08-07:
+1 having the same issue

richard-scott, 2021-08-09:
Have you tried this?

```
brew install dfu-util
```

mkintzel, 2021-08-12:
Thank you @richard-scott  that solved my problem!!  Now to get a model setup and back to flying!  Love the new Edge interface!

CoderElectronics, 2021-08-14:
> Have you tried this?
> 
> ```
> brew install dfu-util
> ```

That's the perfect solution for now, i'm compiling new dfu-util binaries as we speak that should fix that

closed 2021-09-06 by CoderElectronics

lerni, 2022-04-20:
On OSX (as originally reported) installing dfu-util per brew is still needed with latest (#16fb26b3562e653abe9d09c6ecc843a257d79c7b) as of 20th of april 22