EdgeTX/flasher #29: Change software source

norulers, 2021-09-28:
In China, due to network reasons, firmware and SD card files cannot be updated online most of the time .

How to set up a mirror server by yourself (only the software part, and the server hardware is ready)

CoderElectronics, 2021-10-01:
So the firmware is actually pulled directly from Github, you may need to download the files and flash manually. Or use a VPN to somewhere that GH is accessible and then try the app

pfeerick, 2021-10-07:
If you have the server resources to build yourself, you can look use the following to see how the firmware is being build currently (the actual build script, and the github action that runs it).
- https://github.com/EdgeTX/edgetx/blob/main/tools/build-gh.sh
- https://github.com/EdgeTX/edgetx/blob/main/.github/workflows/actions.yml#L69-L120

You can basically use the build script above in the right docker environment to generate any firmware you want, and in the language you want (where supported). Once you have the files, it's then just a matter of hosting them somewhere if you want to make them available in your region ;)

closed 2021-11-18 by CoderElectronics