EdgeTX/flasher #34: Radio screen doesn't shows up

jaggie-xd, 2021-11-14:
Got this error while flashing with DFU and radio seems to be dead somehow.
how should I recover  my radio back!

Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash "
Downloading element to address = 0x08000000, size = 13297

[image]
40

CoderElectronics, 2021-11-18:
Hmm, that's very strange, This looks like a radio-level issue but i have a few ideas. 

1. Make sure your radio is powered off and plugged into USB.
2. Open debugging options dropdown in the flasher screen and check "Force Unprotect Before Flashing.

If that doesn't work, i would try saving the firmware file and flashing via DFU with companion.

Decones80, 2021-12-27:
I have similar issue here, got this corrupted message.....then I rollback to OpenTx using Companion and tried agian, now the error is related to vendor error....wtf. My Tx model is Radiomaster  TX16S, here is the error message:

_Determining device status...
DFU state(10) = dfuERROR, status(11) = iString indicates a vendor specific error
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash "
Downloading element to address = 0x08000000, size = 1131512_

I will rollback it again and stick to OpenTx until it turn allright.... :-/