EdgeTX/flasher #35: Version still not being shown on Windows builds

pfeerick, 2021-11-25:
As has been pointed out a few times now, and was also the reason for https://github.com/EdgeTX/flasher/pull/27, version numbers are still not being shown on Windows builds. Visible here, also on the screenshot Risto posted in Discord when talking about the wizard. 

[image]

CoderElectronics, 2021-11-26:
Let me look at this

Prasanna-icefire, 2022-02-25:
Same issue.....I tried checking the linux deb app, the issue persists...

pfeerick, 2022-02-26:
With the latest release? I do get it on Windows now. 

[image]

Prasanna-icefire, 2022-02-26:
[image]

This is the issue faced....

pfeerick, 2022-02-26:
Ah, ok, that is different problem to the one this issue was in relation to - it was only no version number was being shown on the windows builds (which is resolved now, as the github hash is being shown now - although that still needs improving on), not that no downloads were available. I'm seeing that also now that I try here. This could be a github issue, or an issue with flasher, not sure yet.

richard-scott, 2022-02-26:
I get this on Ubuntu too ;-(

CoderElectronics, 2022-04-18:
This is all fixed, closing now, if you are still missing it update to the latest GH release and it should be there, please reopen if this is incorrect.

closed 2022-04-18 by CoderElectronics