EdgeTX/flasher #44: Error when trying to flash on a Mac

PaulDokken, 2022-05-30:
I have an M1 mac and am trying to run the flasher and the companion.  When i run the flasher i got the following error.
[image]

Also the companion does not recognize my radio.

jeremyhu, 2022-07-22:
This is a pretty serious issue that makes the flasher completely unusable on macOS for most users.

The issue here is that the dfu-util executable has a link against a library that it depends on (libusb-1.0.0.dylib).  At build time, the library that was linked against contained a dylib id of `/usr/local/opt/libusb/lib/libusb-1.0.0.dylib` which is where the dynamic linker / loader (dyld) looks for it on the system at runtime.

This library is located within the edgetx-flasher.app app bundle.  The build system should be using `@loader_path` relative links to resolve these.  There's also no reason to distribute these win64 and linux binaries as part of the darwin flasher app, but that's not the core problem here.

Also note that the release contains two separate copies of the dylib.  Presumably these were at one point hard or symlinks.  There's also a static archive version and a libtool .la which are useless to end users:

```
$ ls -al
total 980
drwxr-xr-x 11 jeremy admin    352 Apr 18 08:02 .
drwxr-xr-x  5 jeremy admin    160 Apr 18 08:02 ..
-rwxr-xr-x  1 jeremy admin  19612 Apr 18 07:59 dfu-prefix
-rwxr-xr-x  1 jeremy admin  19604 Apr 18 07:59 dfu-suffix
-rwxr-xr-x  1 jeremy admin  57756 Apr 18 07:59 dfu-util
-rwxr-xr-x  1 jeremy admin 124856 Apr 18 07:59 libusb-1.0.0.dylib
-rw-r--r--  1 jeremy admin 405136 Apr 18 07:59 libusb-1.0.a
-rwxr-xr-x  1 jeremy admin 124856 Apr 18 07:59 libusb-1.0.dylib
-rwxr-xr-x  1 jeremy admin    975 Apr 18 07:59 libusb-1.0.la
-rwxr-xr-x  1 jeremy admin 206188 Apr 18 07:59 lsusb
-rwxr-xr-x  1 jeremy admin  28368 Apr 18 07:59 usbhid-dump
```

Ok... so for fixing things...

```
cd /Applications/edgetx-flasher.app/Contents/src/support/dfu-util/darwin 
rm libusb-1.0.a libusb-1.0.la libusb-1.0.dylib
install_name_tool -id @loader_path/libusb-1.0.0.dylib libusb-1.0.0.dylib
install_name_tool -change /Users/tormod/dfu-util/libusb-git/../build/lib/libusb-1.0.0.dylib @loader_path/libusb-1.0.0.dylib lsusb
install_name_tool -change /Users/tormod/dfu-util/libusb-git/../build/lib/libusb-1.0.0.dylib @loader_path/libusb-1.0.0.dylib usbhid-dump 
install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @loader_path/libusb-1.0.0.dylib dfu-prefix
install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @loader_path/libusb-1.0.0.dylib dfu-suffix 
install_name_tool -change /usr/local/opt/libusb/lib/libusb-1.0.0.dylib @loader_path/libusb-1.0.0.dylib dfu-util 
```

richard-scott, 2022-07-25:
I've had to run this to get it to work:

``` bash
brew install dfu-util
```

bigtwin, 2022-07-28:
I have a highly similar symptom while trying to use _edgetx-flasher_ on my Intel-based _Mac OSX Mojave_ to work with my _Taranis X9 Lite_. Unfortunately I neither understand @jeremyhu analysis nor what is ment by @richard-scott contribution. I cannot judge whether this is related at all:

Waiting for dfu-util...dyld: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
Referenced from: /Applications/edgetx-flasher.app/Contents/src/support/dfu-util/darwin/dfu-util
Reason: **image not found**

Coming from OTX, this is my very first start with EdgeTX and I am not sure how to handle the situation I am in. Is it worth for me proceeding with EdgeTX now (do I need edgetx-flasher?), or better to wait for e.g. next release?