EdgeTX/flasher #45: Problem with "Setup SD Card"

sergiobellini, 2022-06-08:
Hi,
when click on "Setup SD Card" no Voicepack and no Lua scripts are shown, and no Disk into selectbox.

In loggin:
[updateConfig.message.error][2022-06-08T13:30:40.338Z] Error: Request failed with status code 404

Thanks

munzli, 2022-06-21:
same problem. 
tried linux deb and appimage

Potatotatotato, 2022-06-27:
Me,too

jawz101, 2022-07-28:
On Windows I get an error in the event log

```
The application-specific permission settings do not grant Local Activation permission for the COM Server application with CLSID 
{2593F8B9-4EAF-457C-B68A-50F6B8EA6B54}
 and APPID 
{15C20B67-12E7-4BB6-92BB-7AFF07997402}
 to the user computername\username SID (S-1-5-21-2752922228-1345440906-3538473472-1001) from address LocalHost (Using LRPC) running in the application container Unavailable SID (Unavailable). This security permission can be modified using the Component Services administrative tool.
```

xxz3, 2022-07-28:
+1 on MacOS Big Sur 11.6.5 - edgetx flasher 16fb26b

brunohenriquy, 2022-07-28:
Same here, MacOS and Windows 11.

brunohenriquy, 2022-07-28:
https://docs.microsoft.com/en-us/answers/questions/281617/the-application-specific-permission-settings-do-no-1.html
Tried this, but don't work!

PR #46 (2022-08-11): fix: Update names of sdcard zips
Needed after https://github.com/EdgeTX/edgetx-sdcard/pull/64

This is really only a quickfix... ideally the `sdcard.json` in the repo would be used since that has all the names and targets. 

Resolves #45
closed 2022-08-11 by CoderElectronics