EdgeTX/flasher #47: All languages selected.

duffmaxwell, 2022-09-24:
New to this but tried using the flasher to flash my jumper t light v1 and when I went to click English for the audio file it selected all of the checkboxes instead of just English. It did this no matter what I clicked. When I tried to flash anyway it got stuck at retrieving audio file.

duffmaxwell, 2022-09-24:
on windows fyi

Fahersto, 2022-10-29:
I'm facing the same issue on FrSky QX7 and Windows 10

petewil0, 2022-11-06:
I'm seeing this too on 629b68c, on Win 10, using a Taranis QX7 (frsky).  It selects all the languages, then when I click "Write to SD card", I get a confirmation dialog.  I press "continue" on the confirmation dialog, and Edge TX Flasher hangs.  So, basically I can't flash the SD card on windows, which stops my migration to EdgeTX in its tracks.

I'd like to request that you label some older known working builds - Ideally I'd install an older build instead of "latest" which might not have had this problem introduced.

Fahersto, 2022-11-06:
For the time being you can just not select any language pack and then manually copy them into the language folder on the sd card. This is how I did it atleast.

petewil0, 2022-11-06:
I can confirm that following the manual instructions on the github page for EdgeTX worked for me.

wawanlost, 2022-12-09:
same problem on radiomaster tx16s, and also doesn't solve for long time (failed),audio file remains empty..

IoakeimSogiakas, 2022-12-15:
I also have the same isssue using the radiomaster zorro. Also changing lua scripts does nothing